Rat-OS/RatOS-configurator #2: feature request:  calibration prints and calibration of filament

Trackhe, 2022-10-26:
feature request:  calibration prints and calibration of filament

i saw the next available features in the index and i m curios to see this. i tried the wifi task already and its very cool.
so my second thought was that creating own filament pre/profiles would be a good extention at the end. and calibration before too. at the end the use can print the RatOS / Mainsail Logo or the 3dbenchy ^^.

miklschmidt, 2022-11-07:
There'll be a lot of features in the future, but the focus is on solving real problems right now, and there are quite a few of those that take precedence:

1) Control board / toolboard flashing wizard (this is done and will ship next release)
2) Kinematics wizard to automatically set motor directions based on a few simple steps.
3) Full blown klipper config wizard (the endgame).

Filament profiles would come after those.

The configurator cannot be hosted remotely, it has to have access to scripts and commands on the pi for the majority of the tasks. The size is fixable, but i'm currently focused on shipping features, so minor optimizations like this will have to come later.

I'll leave this open for now, if someone wants to work on it.

Trackhe, 2022-11-07:
Sounds very good.