Rat-OS/RatOS-configurator #4: Feature Request: Disable Wifi button in Configurator.

Mitsuma, 2023-01-29:
RatOS opens a Wifi hotspot even if you use a wired connection.
Simplest solution, if auto detection is not an option, would just to add a button that disables the Wifi interface in the configurator.
Should either be in the Skip button of the WIFI SETUP part or an extra button there.

manueldrama, 2023-12-13:
I am looking for the same solution. I guess we can create a macro for a shell script to on off hotspot with a bash script code.

hapklaar, 2024-03-08:
Same. I don't want this extra hotspot in my house when I don't use it.

dohlin, 2024-04-23:
This would be very nice.  For now I just connected my Pi to my wireless, but since it's been hardwired since minute 1 it feels very redundant and unncessary!

romarysonrier, 2024-10-20:
This feature is a must have, I don't want my office or home wifi band being poluated by useless hot spot. It should even not start AT boot once the setting is done.

emersonsc, 2024-11-18:
PLEASE ADD THIS! I dont like a printer being wide open like this... quite the security hole

foundnoname, 2025-04-05:
For anyone else bothered by this, you can manually disable the AP by logging in via SSH and then run `systemctl disable --now hostapd`