Rat-OS/RatOS-configurator #12: Not reproduced yet: SSID name can cause wifi wizard config issues

miklschmidt, 2023-05-13:
User report:
> My Wifi name is \o/ but it is shown as \x5co/. When I try to connect to it via the wizard of ratos 2, I can put in a password, but connecting does not work. I changed the Wifi to something with only letters and now it works, but wanted to report it nonetheless. 

[image]

funtimeerror, 2025-01-20:
@miklschmidt this was brought up this morning in Facebook in the official Ratrig group, https://www.facebook.com/share/p/1Chc17J8uQ/

A ' broke the SSID prevented the password from encrypting correctly. Please assign this one to me, I'll review and submit a PR :)

miklschmidt, 2025-01-21:
@funtimeerror Awesome, much appreciated!

funtimeerror, 2025-01-21:
PR #65 has been submitted

closed 2025-01-21 by miklschmidt