Rat-OS/RatOS-configurator #27: Prusa MK3S: display_rs_pin is not a valid pin name on mcu

kronflux, 2024-11-04:
Using RatOS-v2.1.0-RC2, Prusa MK3S does not generate configuration properly.
It is missing definitions for the display and filament sensor. It also assumes, regardless of having selected no accelerometer, that there is an adxl345 controlboard. but tries to use them anyway.
Reverting to RatOS-v2.0.2 does not have these issues, after a lot of tinkering and investigating, this appears to be an issue with the configurator.

kronflux, 2024-11-07:
This may be more info than needed, and may need to be added to another issue on one of the other repos, but for reference sake, upon initial installation, if you go to ratos.local, you get a bed_mesh error (I assume due to not having run configurator yet)
[moonraker.log](https://github.com/user-attachments/files/17668940/moonraker.log)
[klippy.log](https://github.com/user-attachments/files/17668941/klippy.log)

Upon running configurator, then going back to ratos.local, I get "display_rs_pin" is not  a valid pin name on mcu "mcu"
[moonraker_after_configurator.log](https://github.com/user-attachments/files/17669027/moonraker_after_configurator.log)
[klippy_after_confiurator.log](https://github.com/user-attachments/files/17669028/klippy_after_confiurator.log)

Here is the Debug Zip:
[ratos-debug.zip](https://github.com/user-attachments/files/17669032/ratos-debug.zip)

miklschmidt, 2024-11-08:
The reason is that the mk3 implementation uses non-standard pin aliases that get stripped. I have removed the display and filament sensor includes for now, as we currently don't have a solution for those. RatOS does have [a way to manually set up filament sensors](https://github.com/HelgeKeck/RatOS/blob/documentation_v2.1/site/docs/configuration/filament_sensors.md) however.

That would result in the following, using a Prusa Einsy:

```properties
[filament_switch_sensor toolhead_filament_sensor_t0]
pause_on_runout: False
event_delay: 3.0
pause_delay: 0.01
switch_pin: ^!PK0
runout_gcode:
    _ON_TOOLHEAD_FILAMENT_SENSOR_RUNOUT TOOLHEAD=0
insert_gcode:
    _ON_TOOLHEAD_FILAMENT_SENSOR_INSERT TOOLHEAD=0
```

For the display, you'll manually have to add the following to printer.cfg:
```properties
[display]
lcd_type: hd44780
rs_pin: PD5
e_pin: PF7
d4_pin: PF5
d5_pin: PG4
d6_pin: PH7
d7_pin: PG3
encoder_pins: ^PJ1,^PJ2
click_pin: ^!PH6

[output_pin BEEPER_pin]
pin: PH2
pwm: True
value: 0
shutdown_value:0
cycle_time: 0.001
scale: 1000

[menu __filament __load]
type: command
name: Load Filament
gcode:
    LOAD_FILAMENT

[menu __filament __unload]
type: command
name: Unload Filament
gcode:
    UNLOAD_FILAMENT
 ```
 
 Note: the above will only work with the Prusa Einsy board.
 
I have added this to the default printer.cfg generated for the mk3s, so users can uncomment them if they're using the Prusa Einsy, and either leave them or modify them for other boards. The configurator should be ready to be updated in 10 minutes. This is the commit with the fixes: https://github.com/Rat-OS/RatOS-configurator/commit/069dee1e54d8d8b1c5b4470a4479e255cd7da662

miklschmidt, 2024-11-08:
I have also fixed the bed_mesh error (it only happens on unconfigured printers, but still annoying)

kronflux, 2024-11-08:
Unfortunately the configurator in this state results in multiple definitions of "PE6" pin, I believe this is due to an invalid pin definition in heater_fan toolhead_cooling_fan, where the pin is set to PE6 as is the tachometer pin.
The pin should be PH5, and the tachometer pin should be PE6.

After resolving that, I get "Section 'fan part_cooling_fan' is not a valid config section"
which seems to correspond to the generated printer.cfg, where 
we've used "fan part_cooling_fan" where previously it was simply "fan" although I'm not positive on that one.
Here is the debug zip:
[ratos-debug (1).zip](https://github.com/user-attachments/files/17674315/ratos-debug.1.zip)

I liked in the stable release of RatOS where it used a separate cfg file for various things, such as display. A modular approach would be ideal for every component, including filament sensor for each printer, display for each printer, etc

kronflux, 2024-11-08:
It's also not selecting the correct definition for the pi, in my configuration I have a pi4 hooked up to the printer via USB. In 2.x it detects this and flashes appropriately, and everything sort of just "works" but in RC2 there seems to be a lot of issues.
Here's what I'm getting:
mcu 'rpi': Starting connect
Loaded MCU 'rpi' 119 commands (?-20240727_132503-fv-az659-741 / gcc: (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110 binutils: (GNU Binutils for Raspbian) 2.35.2)
MCU 'rpi' config: ADC_MAX=4095 CLOCK_FREQ=50000000 MCU=linux PCA9685_MAX=4096 PWM_MAX=32768 STATS_SUMSQ_BASE=256
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (600 moves)
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 135, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 742, in _connect
    config_params = self._send_get_config()
  File "/home/pi/klipper/klippy/mcu.py", line 728, in _send_get_config
    raise error("Can not update MCU '%s' config as it is shutdown" % (
mcu.error: Can not update MCU 'rpi' config as it is shutdown
Can not update MCU 'rpi' config as it is shutdown
[ratos-debug (2).zip](https://github.com/user-attachments/files/17674365/ratos-debug.2.zip)

miklschmidt, 2024-11-24:
@kronflux i'm sorry i haven't had time to look at this as we've been working on a new post-processor, which ended up having some rather complicated implications. I'll try and get this sorted once it's been merged. 

I don't think there's anything wrong with the rpi config (there isn't anything to select), that error you're getting is likely a symptom of another error. Will investigate further when i get back to this!

Regarding the tach pin, the pin definition for the board hasn't changed since 2.0 so there's likely some more unorthodox stuff going on with the old 2.0 config to make it work that i've missed.

kronflux, 2024-12-04:
No worries! It was easy enough to work around, just more hoping to get a perfect out of the box config for other users :)

kronflux, 2024-12-26:
Just as an update, I did a fresh install of the latest RC3 artifact from github actions, did the standard setup tasks (connect to it via wifi, configure wifi network, update ratos, ratos-configurator, then update all, then run configurator for my setup)

automatic flashing is failing:

```
Error
Could not compile firmware for Einsy: Could not find compiled firmware at /home/pi/klipper/out/klipper.hex 

 /home/pi/klipper /home/pi/ratos-configurator/app
  Creating symbolic link out/board
Loaded configuration '/home/pi/klipper/.config'
Configuration saved to '/home/pi/klipper/.config'
  Creating symbolic link out/board
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_hx71x.o
  Compiling out/src/sensor_ads1220.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/adc.o
  Compiling out/src/avr/spi.o
  Compiling out/src/avr/i2c.o
  Compiling out/src/avr/hard_pwm.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/serial.o
  Compiling out/src/generic/serial_irq.o
  Building out/compile_time_request.o
Version: v0.12.0-348-geeb2678ec
  Linking out/klipper.elf
  Creating hex file out/klipper.elf.hex
/home/pi/ratos-configurator/app
```

but I have a previously flashed copy, so I move on.

if I select "None" as the accelerometer, I get:

```
Error
Cannot translate accelerometer id to Klipper accel sensor name
```

if I do not select anything, it seems to use:
accel_chip_x: adxl345 controlboard
accel_chip_y: adxl345 controlboard

Regardless of what I configure here, I get the following when I complete the setup:

```
Klipper reports: ERROR

Internal error during connect: module 'extras.resonance_tester' has no attribute 'SweepingVibrationsTestGenerator'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

[ratos-debug.zip](https://github.com/user-attachments/files/18251573/ratos-debug.zip)

miklschmidt, 2024-12-26:
@kronflux appreciate it! Fixed the first 2 in the development branch, the last one just requires you refresh the update manager and update Klipper (they changed the resonance testing).

miklschmidt, 2024-12-26:
> [fan part_cooling_fan]
cycle_time: 0.001

Are you sure? What happens if you leave it in?

> It is also worth noting that the Display and Filament Sensor configurations are commented out by default, however all MK3S printers would come with these, so there's not much purpose to commenting them out.

I had to comment those out because they're not natively supported by RatOS so those pin aliases don't exist on other boards.

kronflux, 2024-12-26:
> > [fan part_cooling_fan]
> > cycle_time: 0.001
> 
> Are you sure? What happens if you leave it in?
> 

`Section 'fan part_cooling_fan' is not a valid config section`

I am also bashing my head against the wall trying to figure out this pin definition issue. 

No matter what I change, upon reboot, the RatOS.cfg gets overwritten with this:

`# Hotend cooling fan
[heater_fan toolhead_cooling_fan]
heater: extruder
# 4-pin fan connected to 4-pin header on Einsy - digital pwm
pin: PE6
cycle_time:  0.00004
tachometer_pin: ^PE6
tachometer_poll_interval: 0.0005`

it really wants that pin to be PE6, but it should be PH5. I've been searching through every config, definition, template, etc. I cannot for the life of me figure out where it's getting PE6 there 🤣

david-antos, 2024-12-26:
I have corrections for some of the issues by @kronflux. Taking up-to-date RatOS versions into account, I had to do following overrides for MK3S:

[heater_fan toolhead_cooling_fan]
pin: PH5
cycle_time: 0.01
# pin was wrong
[fan]
cycle_time: 0.01
# it should be 100 Hz, but the hack in the generated printer.cfg sets it to 0.001, which is 1 kHz. 100 Hz worked in RatOS 2.0 and reported correct tacho values, so 100 Hz is most likely correct.

Regarding sensorless homing, I believe this should be necessary to correct as well:
[tmc2130 stepper_x]
diag1_pin: ^PK2

[tmc2130 stepper_y]
diag1_pin: ^PK7
*nevertheless* I haven't managed to get sensorless homing to work. Regardless of homing_speed set for the steppers, it just crashes full speed.

@kronflux, I'd suggest there is little point to reinstall RatOS from scratch, support for MK3S is completely messy in the release candidate, it is necessary to upgrade the installation *before* you even attempt configuring it via the web config.

kronflux, 2024-12-26:
> I have corrections for some of the issues by @kronflux. Taking up-to-date RatOS versions into account, I had to do following overrides for MK3S:
>
> ```
> [heater_fan toolhead_cooling_fan]
> pin: PH5 [...]

Appreciate the input! That being said, I am aware of the present state, I am not dependent on it, as I do have another SD card with a working setup, this thread is meant to help give input on what needs fixing for Mk3S

miklschmidt, 2024-12-26:
> `Section 'fan part_cooling_fan' is not a valid config section`

Right it's just `fan`. Must be hard coded in the template, will fix.
 
> I am also bashing my head against the wall trying to figure out this pin definition issue.
> 
> No matter what I change, upon reboot, the RatOS.cfg gets overwritten with this:
> 

Read the first line in RatOS.cfg.

> ```
> # Hotend cooling fan
> [heater_fan toolhead_cooling_fan]
> heater: extruder
> # 4-pin fan connected to 4-pin header on Einsy - digital pwm
> pin: PE6
> cycle_time:  0.00004
> tachometer_pin: ^PE6
> tachometer_poll_interval: 0.0005
> ```
> 
> it really wants that pin to be PE6, but it should be PH5. I've been searching through every config, definition, template, etc. I cannot for the life of me figure out where it's getting PE6 there 🤣
> 
> EDIT: Just for the heck of it, I just did a fresh install and update, prior to configuration, it reports:
> 
> ```
> Existing command 'TEST_RESONANCES' not found in RatOS override

You did not complete the update, we merged the configuration into the configurator before christmas, all current images need to migrate past that point, and it's a bit tricky as moonraker and klipper need to be restarted to pick up the filesystem and config changes. Read this: https://os.ratrig.com/docs/installation#updating-ratos

This will not be an issue on release, however the process outlined there has always been the recommended way to update, RatOS can prepare for thirdparty changes if it isn't updated separately. We may replace the update manager in a future version.

> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Printer is halted
> ```
> 
> and the RatOS Configuration repository shows as invalid. I tried configuring, and automated flash still failed with the same error.

See above.

> 
> Selecting "None" for X and Y accelerometer gives us a lovely:
> 
> ```
> Error
> Cannot translate accelerometer id to Klipper accel sensor name
> ```
> 

As i mentioned, this fix is on the development branch that is currently undergoing testing, it has not been merged yet.

> so I had to select Wired to Host Computer.
> 
> After this, we're back to the Mainsail page, and have the error:
> 
> ```
> Internal error during connect: module 'extras.resonance_tester' has no attribute 'SweepingVibrationsTestGenerator'
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Printer is halted
> ```
> 

Same thing again.

> trying a hard recovery on the RatOS Configuration repository gives a "Cannot call rmtree on a smbolic link" (I imagine this is expected, but it was worth a shot)
> 

Same thing.

> Here's the debug zip for reference [ratos-debug (1).zip](https://github.com/user-attachments/files/18252803/ratos-debug.1.zip)
> 
> EDIT2: ahh it seems the RC3 builds are still using the default branch of Configurator rather than the development branch, so the most recent commits weren't present. unfortunately manually switching the branch, it seems to break the configurator entirely (502 Bad Gateway) so I can't really test it without some guidance :P

You cannot (and shouldn't) run main branches on a pi without additional setup, the configurator needs to be compiled, we have deployment branches for that. You should not run the dev-deployment branch, it's for development and breaks regularly.

kronflux, 2024-12-26:
> > `Section 'fan part_cooling_fan' is not a valid config section`
> 
> Right it's just `fan`. Must be hard coded in the template, will fix.
> 
Awesome, thank you :)

> You cannot (and shouldn't) run main branches on a pi without additional setup, the configurator needs to be compiled, we have deployment branches for that. You should not run the dev-deployment branch, it's for development and breaks regularly.
>
Yeah, while I am aware of this, the intent is me attempting to test and identify issues for upcoming releases, so we can get this printer fully supported in the next major release. I understand this is not meant for daily use, but it does pose a problem where the action builds still use the mainline branches. Oh well

> You did not complete the update, we merged the configuration into the configurator before christmas, all current images need to migrate past that point, and it's a bit tricky as moonraker and [...]
I did actually follow the instructions to a tee, but I imagine it did not succeed due to the branch differences.. I will give it another try at some point. But I may be better off waiting for the official RC3 build rather than using github action builds..

miklschmidt, 2024-12-26:
> Yeah, while I am aware of this, the intent is me attempting to test and identify issues for upcoming releases, so we can get this printer fully supported in the next major release. I understand this [...]

In that case: 
```
ssh pi@ratos.local
cd ratos-configurator
git fetch origin
git checkout -b dev-deployment origin/dev-deployment
ratos doctor
```
followed by refreshing the update manager panel, then updating the remaining packages. Be sure to go back to `v2.1.x-deployment` once testing is concluded.

> I did actually follow the instructions to a tee, but I imagine it did not succeed due to the branch differences..

Branch differences? You mean you followed the instructions *after* checking out the `development` branch? If you followed the instructions from the start, you would've restarted moonraker, refreshed the update manager and then updated the remaining packages and you wouldn't have that error.

kronflux, 2024-12-27:
Ahh I see! I did indeed switch branches first, that's probably where I went wrong, I'll give this a shot again once my current print is complete and follow those instructions so I can do some further testing and get back to you.
Thanks for the prompt replies :)

miklschmidt, 2024-12-27:
@kronflux Thank you for helping me get this thing working again, and thank you for the sponsorship! :)

miklschmidt, 2024-12-27:
Looks like you're the first to use (or at least report) a dedicated 4 pin header for the toolhead fan 😂 That one should be fixed as well now!

kronflux, 2024-12-27:
> @kronflux Thank you for helping me get this thing working again, and thank you for the sponsorship! :)

My pleasure! The work done on RatOS is amazing, and I hope to see it support a lot more printers in the future!

Okay, I may have missed a step, I'm not sure, but I went through the standard setup and update process with the latest RC3 build(perhaps I should be using the RC2 build?) and followed these instructions to switch branch and run RatOS Doctor
> ```
> # IF YOU JUST CAME ACROSS THIS POST, DO NOT RUN THIS.
> ssh pi@ratos.local
> cd ratos-configurator
> git fetch origin
> git checkout -b dev-deployment origin/dev-deployment
> ratos doctor
> ```
then proceeded with the setup wizard. The automatic flashing still seems to be failing, I'm not sure if the branch simply didn't fully switch, or whether the fix was unsuccessful.
Here is the debug zip from prior to continuing the setup:
[ratos-debug (2).zip](https://github.com/user-attachments/files/18256860/ratos-debug.2.zip)
as I proceed, I do still also get "Accelerometer type could not be determined for none" when I select None for Accelerometers.

But! The duplicate pin issue is now fixed, we don't get the "Section 'fan part_cooling_fan' is not a valid config section", and no other errors once configured.

Definitely making progress! 😁

Here's a debug zip from after everything is configured as well:
[ratos-debug (3).zip](https://github.com/user-attachments/files/18256884/ratos-debug.3.zip)

miklschmidt, 2024-12-27:
@kronflux gah! Should be fixed when this is deployed in 5 minutes https://github.com/Rat-OS/RatOS-configurator/actions/runs/12511314581

kronflux, 2024-12-27:
Excellent! That fixed the Accelerometer type None error message, and the firmware flashing issue!
Unfortunately we are now getting:
`Unknown config object 'adxl345 controlboard'`
So by selecting None, it seems to default to adxl345
[ratos-debug.zip](https://github.com/user-attachments/files/18258901/ratos-debug.zip)

The bigger downside here being that the configuration for it is being placed in RatOS.cfg, and I'm unsure how to "unset" values using printer.cfg overrides.
So my workaround as like before is to select the "Wired To Host Computer" option.

with the default "Physical Endstop" options for X and Y Endstops, we get:
`Endstop x still triggered after retract`
since the Prusa MK3S does not have endstops and uses virtual endstops, we need to set this to Sensorless Homing.
However, this gives the error:
`tmc virtual endstop requires diag pin config`
so it would seem something is misconfigured either with virtual endstops or our diag pin definitions.

This may?? be caused by the board definitions' diag pins being set to PK2, PK7 and PK6, where these under normal circumstances should be inverted ( !PK2, !PK7, !PK6 )
Or the definitions are not being used correctly.

If I define these manually in printer.cfg as such:
```
[tmc2130 stepper_x]
diag1_pin: !PK2

[tmc2130 stepper_y]
diag1_pin: !PK7

[tmc2130 stepper_z]
diag1_pin: !PK6
```
It does seem to resolve the issue, and I can home normally.

miklschmidt, 2024-12-27:
It shouldn't have emitted the `[resonance_tester]` section.. Added a test and fixed, online in 5 minutes :)

miklschmidt, 2024-12-27:
@kronflux I've set the mk3s to default to sensorless homing using the pretuned configuration (which can be overridden in the `sensorless-homing-*.cfg` files). Added tests to ensure they're emitted correctly. To reset your defaults either reselect the printer (select a different one, than mk3 again and pick "start fresh"), or set both x and y to sensorless in your hardware configurator.

Fix is online in 5 minutes :)

david-antos, 2024-12-27:
I'd be happy to put my MK3S to the party. I've switched to devel for the purpose of testing (as described above):
RatOS Core: Configurator & Configuration
v0.0.0-411-g96857c7c-inferred
(no other upgrades are indicated by the upgrade manager).

Re-run the wizard for MK3S starting fresh with sensorless homing. I still need to modify printer.cfg adding diag1_pins for the drivers as discussed above. 
[ratos-debug.zip](https://github.com/user-attachments/files/18260594/ratos-debug.zip), all relevant changes are in printer.cfg marked as CORRECTIONS.

Issue 1:
Sensorless homing seems to ignore homing_speed and just crashes the bed into the frame high speed, making several very ugly skips when it crashes (I don't want to do *that* often to my printer ;).)

Issue 2:
The special sensorless homing setup for MK3S is included in RatOS.cfg sets homing speeds to 25. Those get overwritten by printer.cfg, as its default configuration lists stepper parameters to choose and sets it back to 50. (I've changed those to 10 in printer.cfg, anyway, it shoots the bed way way way faster than that.)

Issue 3:
@miklschmidt asked in this issue whether fan cycle_time 0.001 really matters: I can confirm it does, at least tacho reading out of the fans were completely messed up (reporting crazy values like 16000 when the fan is 50%), and it seems the fans don't go full blast when set to 100%. I'd suggest changing those to 100 Hz, i.e. 0.01, is necessary.

miklschmidt, 2024-12-27:
> I'd be happy to put my MK3S to the party. I've switched to devel for the purpose of testing (as described above): RatOS Core: Configurator & Configuration v0.0.0-411-g96857c7c-inferred (no other [...]

There was a type error so the fixes didn't get deployed, please try again.

> 
> Issue 1: Sensorless homing seems to ignore homing_speed and just crashes the bed into the frame high speed, making several very ugly skips when it crashes (I don't want to do _that_ often to my printer ;).)
> 

See above. If it still crashes, the "pretuned" configuration apparently isn't as pretuned as the original author claimed and you'll have to tune the values in sensorless-homing-*.cfg (instructions are in the files, and there's a reference to them in printer.cfg). Sensorless homing is supposed to home relatively fast, too slow and it won't trigger reliably.

> Issue 2: The special sensorless homing setup for MK3S is included in RatOS.cfg sets homing speeds to 25. Those get overwritten by printer.cfg, as its default configuration lists stepper parameters [...]

neither 25 or 50 isn't "fast" or problematic for sensorless homing, those should work fine. I would not use 10 mm/s for sensorless homing.

> Issue 3: @miklschmidt asked in this issue whether fan cycle_time 0.001 really matters: I can confirm it does, at least tacho reading out of the fans were completely messed up (reporting crazy values [...]

I misunderstood your previous comment, i didn't realize there was an extra zero until now. Fixed.

david-antos, 2024-12-27:
> See above. If it still crashes, the "pretuned" configuration apparently isn't as pretuned as the original author claimed and you'll have to tune the values in sensorless-homing-*.cfg (instructions [...]

I wouldn't doubt the values, as I hapilly used RatOS 2.0 values for quite some time, homing worked out of the box and has been rock solid since. The configuration in sensorless-homing-*.cfg and driver parameters are the same as it used to be, except for breaking driver_SGT, which is easily corrected with:
```
[tmc2130 stepper_x]
driver_SGT: 5

[tmc2130 stepper_y]
driver_SGT: 5
```
And I'm happy to report that those values work gently and quietly. And if those were shown to work in 2.0, I'd kindly suggest to make them the default in the MK3S configuration again as they used to be. Driver_SGT is set to 5 in the stepper config, and changed to 0 in the RatOS/printers/prusa-mk3s/sensorless-homing-tmc2130.cfg later.

> > Issue 2: The special sensorless homing setup for MK3S is included in RatOS.cfg sets homing speeds to 25. Those get overwritten by printer.cfg, as its default configuration lists stepper parameters to choose and sets it back to 50. (I've changed those to 10 in printer.cfg, anyway, it shoots the bed way way way faster than that.)
> 
> neither 25 or 50 is "fast" or problematic for sensorless homing, those should work fine. I would not use 10 mm/s for sensorless homing. The reason it fails is likely just because the fixes failed to deploy.

I've tested both 25 and 50 and I fully agree that both 25 and 50 works completely fine, and both seem to work well provided that driver_SGT is 5. 50 is a good default.

> I misunderstood your previous comment, i didn't realize there was an extra zero until now. Fixed.

Oh, I see. Thank you!

In that case, driver_SGT is the only override necessary to make MK3S work with RatOS 2.1, at least talking about the devel branch, that's excellent.

Should we keep the generated configs and switch ratos-configurator back to origin/v2.1.x-deployment? Or would that break things?

miklschmidt, 2024-12-27:
> In that case, driver_SGT is the only override necessary to make MK3S work with RatOS 2.1, at least talking about the devel branch, that's excellent.

Awesome. I didn't realize those SGT values were in the stepper tuning presets, that's a mistake as those values are dependent on the entire mechanical system of the machine and not just the stepper driver settings, they were probably copy pasted when adding the tuning presets for the prusa machines. This has now been fixed (i've moved the value to the mk3 specific sensorless-homing-tmc2130.cfg file where they belong) and it should now work as intended :)

> Should we keep the generated configs and switch ratos-configurator back to origin/v2.1.x-deployment? Or would that break things?

Switching back to v2.1.x-deployment will regenerate RatOS.cfg so you would be back to "broken" state again. I suggest remaining on dev-deployment and avoid updating further for now. I will ping you guys here once development has been merged and you can move back to v2.1.x-deployment :+1:

david-antos, 2024-12-27:
Lovely.

And if I may suggest regarding sensorless-homing-x.cfg and -y, according to https://www.analog.com/en/products/TMC2130.html 2130 datasheet, SGT indeed goes from -64 to 63, but 0 is an "indifferent value" (datasheet pg. 62, no explanation what that means is given, though).
And as we know 0 behaves very strangely (speeding in the opposite direction and crashing violently), it may not be the best idea to suggest people should start with SGT 0, as the commented out value indicates.

kronflux, 2024-12-27:
I just updated and re-ran the wizard, getting:
`Existing command 'TEST_RESONANCES' not found in RatOS override`
basically seems to happen if the accelerometer section is missing from printer.cfg, which is why the workaround using the "host computer" option works, just means we have a nonexistent accelerometer configured lol
[ratos-debug (1).zip](https://github.com/user-attachments/files/18261656/ratos-debug.1.zip)

sensorless homing is very weird now, I seem to be getting drastically different results depending on where the print head is currently located.
Attempting to tune it also acts very weird, likely due to the above comment about it? prior to this most recent update, sensorless homing was working perfectly for me after having set the diag pins. now I don't need to set those, but it's not acting properly.

miklschmidt, 2024-12-27:
> Lovely.
>
> And if I may suggest regarding sensorless-homing-x.cfg and -y, according to https://www.analog.com/en/products/TMC2130.html 2130 datasheet, SGT indeed goes from -64 to 63, but 0 is an [...]

SGT=0 (StallGuard Threshold) just means default sensitivity. A lower value means more sensitive and a higher value means less sensitive. There is no "default" that works on all printers, and the statement "as we know 0 behaves very strangely" only applies to your specific setup at the specific homing current and driver tuning settings you're using. Since the MK3S is created relatively equal, we can assume that holds true for the majority of MK3S out there, but simply changing the belt tension, the stepper current or the driver parameters can change the optimal value for `driver_SGT`. 2130's are especially finicky for sensorless homing as they use a very old version of StallGuard. Similarly 5160's use the same outdated StallGuard version, but have very different current sensing circuitry and may require a very different value from 2130's even on the same printer and board.

> it may not be the best idea to suggest people should start with SGT 0, as the commented out value indicates.

It definitely is best to start with sane defaults (including the default sensitivity of 0), when the actual value is unknown (which it will be on all non mass-produced, printers). These files are generated with the same sane defaults on all printers where the user has opted for sensorless homing, which i actively discourage as it's a long process of trial and error for beginners. Most pick sensorless homing because they think it's easier that wiring endstops, nothing could be further from the truth, also sensorless homing is inherently imprecise as it relies on the back EMF generated when the motor stalls, with up to 4 full steps of positioning error.

miklschmidt, 2024-12-27:
> I just updated and re-ran the wizard, getting: `Existing command 'TEST_RESONANCES' not found in RatOS override` basically seems to happen if the accelerometer section is missing from printer.cfg, [...]

Ah! I'll get that fixed, i thought this was because of old code initially, but that could indeed be the case. I'll ping you once done.

> sensorless homing is very weird now, I seem to be getting drastically different results depending on where the print head is currently located. Attempting to tune it also acts very weird, likely due to the above comment about it?

Just to verify, is the driver_SGT commented out in sensorless-homing-*.cfg as it should be to use the MK3S pretuned defaults?

> 
> prior to this most recent update, sensorless homing was working perfectly for me after having set the diag pins(gracefully crashing, not aggressive). now I don't need to set those, but it's not acting properly, and I can't seem to get a value that doesn't either go entirely the opposite direction and grind, or crashing hard in the correct direction

Going the opposite direction means it's triggering early (it thinks it has reached the end of the axis and is returning to the center). Crashing hard means it's not triggering when it supposed to.

I will need a debug zip to see if everything is set up as expected. If you didn't reselect the printer in the configurator you may still be running old values.

kronflux, 2024-12-27:
oddly enough, I power cycled the entire unit and pi, and it seems fine now, must have been some odd memory thing. 🤨

miklschmidt, 2024-12-27:
@kronflux hmm.. Drivers should be reprogrammed when steppers are enabled. Not sure what that is about.

Anyway, i have fixed the `Existing command 'TEST_RESONANCES' not found in RatOS override`, should be online in 5 minutes.

kronflux, 2024-12-27:
completely unrelated, but is there a RatOS setting for heating the heated bed and hotend simultaneously?
I'm sure I could rewrite the macro, but I figured I should ask if there's already something that exists to do this

miklschmidt, 2024-12-27:
@kronflux Not currently, no. It's doing that to ensure the heat has a chance to spread through the bed and so a potential inductive probe is at a predictable temp. I know the mk3s uses a thin pcb as a heater so it's not as relevant there, but it's in the absolute minority. You can skip preheating the extruder with `variable_preheat_extruder: False` though. Not exactly what you want, but might be close enough.

kronflux, 2024-12-27:
> @kronflux Not currently, no. It's doing that to ensure the heat has a chance to spread through the bed and so a potential inductive probe is at a predictable temp. I know the mk3s uses a thin pcb as [...]

Yeah basically just trying to reduce the time it takes to start a print, but it's easy enough to set up :)

david-antos, 2024-12-27:
> It definitely is best to start with sane defaults (including the default sensitivity of 0), when the actual value is unknown (which it will be on all non mass-produced, printers). These files are [...]

I understand. And I wouldn't use sensorless on anything else than Prusa with the default board/wiring (and definitely not on CoreXY).

kronflux, 2024-12-27:
[ratos-debug (2).zip](https://github.com/user-attachments/files/18262276/ratos-debug.2.zip)
Well, dang, fresh config, and it's crashing hard and mcu shutdown due to stepping at too high a rate or something?

miklschmidt, 2024-12-27:
> [ratos-debug (2).zip](https://github.com/user-attachments/files/18262276/ratos-debug.2.zip) Well, dang, fresh config, and it's crashing hard and mcu shutdown due to stepping at too high a rate or something?

Just to make sure, when you're saying "crashing hard", you mean klipper is throwing an error at you because the slow 8-bit einsy can't keep up with the requested speed, and not sensorless homing, right?

miklschmidt, 2024-12-27:
@kronflux this is related to a different bug i just fixed a few minutes ago, where changing motion settings on a rail would cause some of the implicit state to be overridden with global default values. If you update and go through the configurator again, your microstepping should be reset to 16 (it was at 64 which is definitely too much for an 8-bit board) and interpolation should be enabled (motors won't scream at you).

kronflux, 2024-12-27:
> @kronflux this is related to a different bug i just fixed a few minutes ago, where changing motion settings on a rail would cause some of the implicit state to be overridden with global default values. If you update and go through the configurator again, your microstepping should be reset to 16 (it was at 64 which is definitely too much for an 8-bit board) and interpolation should be enabled (motors won't scream at you because of the low microstep settings).

I'll update now and test

> Just to make sure, when you're saying "crashing hard", you mean klipper is throwing an error at you because the slow 8-bit einsy can't keep up with the requested speed, and not sensorless homing, [...]

not just the error, I mean the actual homing is physically crashing/banging, where prior it was gently stopping.
perhaps the update will fix. I'll test now :)

miklschmidt, 2024-12-27:
@kronflux It's quite likely that the microstepping and interpolation will change the sensorless homing behavior as well yeah, let me know how it goes :)

kronflux, 2024-12-27:
Well, it didn't fix the harsh crash homing, but I'm starting a print now to see if it dies and throws an error again

kronflux, 2024-12-27:
Excellent, no errors during print, but yeah, that homing is aggressive

david-antos, 2024-12-27:
With 16 microsteps+interpolation, sensorless homing works fine, even at speed 50 it isn't aggressive at all (I'll override for 30 or something like that as a matter of personal preference, but I wouldn't consider 50 unreasonable). Power cycled everything just to make sure, still OK. Going for a test print.

BTW, setting variable_purge_after_load to zero is intentional?

miklschmidt, 2024-12-28:
@kronflux please post a debug-zip, want to make sure everything is set as expected.

@david-antos Good to hear! `purge_after_load` is only used during tip forming, which isn't enabled by default and takes a while to tune. The default load macro doesn't use it.

miklschmidt, 2024-12-28:
@david-antos Sorry that's not true, i was thinking of the unload macro. I don't know why 0 is the default, that's what @HelgeKeck requested when he rewrote the filament macros. He may have a better answer :)

kronflux, 2024-12-28:
> @kronflux please post a debug-zip, want to make sure everything is set as expected.
> 
> @david-antos Good to hear! `purge_after_load` is only used during tip forming, which isn't enabled by default and takes a while to tune. The default load macro doesn't use it.

[ratos-debug.zip](https://github.com/user-attachments/files/18265324/ratos-debug.zip)

No issues outside of the harsh homing, where a couple builds ago it was very gentle stops, even if I turned up the homing speed.

kronflux, 2024-12-28:
After doing some testing, I arrived at 3 being the optimal value for driver_SGT.
This also corresponds with most common klipper configurations for the MK3S, including the "gold standard" for premade configurations for the device, by dz0ny.
at 4 or above, it is still too aggressive, and below 2 it behaves unpredictably.
This seems to be the case for both X and Y, although I imagine belt tension could technically interfere with this - however belt tension is standardized to some degree, in the building instructions for these machines, so it is unlikely to be very different between machines, assuming they were built to Prusa standards.

miklschmidt, 2024-12-29:
@kronflux It's very unlikely there'll be one set of sensorless homing settings that will work on all machines. But if @david-antos can confirm 3 works for them as well, we can change it.

david-antos, 2024-12-29:
SGT 3 works fine for me as well, tested for speeds 25 and 50. It seems a bit too gentle at 25, though.

miklschmidt, 2024-12-29:
Excellent. Will change it to 3 and bump homing speed to 50 in the tuned .cfg file just so there's no confusion.

kronflux, 2024-12-30:
@miklschmidt Thank you as well! Happy to help :)

closed 2024-12-30 by kronflux