Rat-OS/RatOS-configurator #29: tachometer_pin is not set correctly

Lino77, 2024-11-05:
# Part cooling fan
# Multiple toolheads with part cooling fans configured
# [fan] will use an unused io pin, proxy m106 settings to active toolhead via macro.
[fan]
pin: rpi:gpio4         # sacrifical part fan, use this to independently control your both toolhead part fans
[fan_generic part_fan_t0]
# 4-pin fan connected to 4-pin header on T0 (SB2209 v1.0 RP2040) - digital pwm
pin: toolboard_t0:gpio15
cycle_time:  0.00004

tachometer_pin: ^gpio12   > right is  tachometer_pin: ^toolboard_t0:gpio12

tachometer_poll_interval: 0.0005
[fan_generic part_fan_t1]
# 4-pin fan connected to 4-pin header on T1 (SB2209 v1.0 RP2040) - digital pwm
pin: toolboard_t1:gpio15
cycle_time:  0.00004

tachometer_pin: ^gpio12  > right is tachometer_pin: ^toolboard_t1:gpio12

tachometer_poll_interval: 0.0005

miklschmidt, 2024-12-27:
This should be fixed now, can you please verify if it's generated correctly for you on the latest version?

Lino77, 2024-12-27:
sorry no 
 #############################################################################################################
### FANS
#############################################################################################################
# Part cooling fan
# Multiple toolheads with part cooling fans configured
# [fan] will use an unused io pin, proxy m106 settings to active toolhead via macro.
[fan]
pin: rpi:gpio4         # sacrifical part fan, use this to independently control your both toolhead part fans
[fan_generic part_fan_t0]
# 4-pin fan connected to 4-pin header on T0 (SB2209 v1.0 RP2040) - digital pwm
pin: toolboard_t0:gpio15
cycle_time:  0.00004
tachometer_pin: ^gpio12
tachometer_poll_interval: 0.0005
[fan_generic part_fan_t1]
# 4-pin fan connected to 4-pin header on T1 (SB2209 v1.0 RP2040) - digital pwm
pin: toolboard_t1:gpio15
cycle_time:  0.00004
tachometer_pin: ^gpio12
tachometer_poll_interval: 0.0005

# Hotend cooling fan
[heater_fan toolhead_cooling_fan]
heater: extruder
# 2-pin fan connected to 2-pin header on Octopus Pro H723 V1.1 - input voltage pwm
pin: PE5
[heater_fan toolhead_cooling_fan_t1]
heater: extruder1
# 2-pin fan connected to 2-pin header on T1 (SB2209 v1.0 RP2040) - input voltage pwm
pin: toolboard_t1:gpio14

# Controller cooling fan
[controller_fan controller_fan]
# 2-pin fan connected to 2-pin header on Octopus Pro H723 V1.1 - input voltage pwm
pin: PD12

miklschmidt, 2024-12-27:
Missed the dedicated 4p toolboard heads. *Now* it should be fixed (well, once the above finishes building and is eventually merged into v2.1.x) :)