Rat-OS/RatOS-configurator #64: Running through configurator on RatOS v2.1.0-RC3 for the first time.

andy83-projects, 2025-01-16:
### What happened

Error
[
  {
    "code": "invalid_type",
    "expected": "object",
    "received": "undefined",
    "path": [
      "hotendFan"
    ],
    "message": "Required"
  },
  {
    "code": "invalid_type",
    "expected": "object",
    "received": "undefined",
    "path": [
      "partFan"
    ],
    "message": "Required"
  }
]

### What did you expect to happen

A working configuration file.  Or at least being able to save and continue and manually edit.

### How to reproduce

Printer: Prusa Mk3s
Board: Manta M5P
Toolhead: N/A

X & Y endstop selections have nothing populated in the list (No Options Found)
Part and Hot End Fan have nothing populated in the list (No Options Found)

Controller Fan has nothing populated in the list (No Options Found)

andy83-projects, 2025-01-16:
any other board selection appears to work correctly, however, I am not using a different board.

m4gn3to, 2025-03-04:
I faced the same problem with a prusa mk3s and a skr 1.4 turbo.